opentx/opentx #4741: Support more than 5 options for Widgets

agibson2, 2017-04-02:
I have a Widget that needs more than 5 options available (I need 7).

I set 7 options setup in main.lua for a Widget but in the GUI it only shows the first 5.

local options = {
  --{ "AltitudeSensor", SOURCE, getFieldInfo("Alt").id },
  --{ "VSpeedSensor", SOURCE, getFieldInfo("VSpd").id },
  { "AltitudeSensor", SOURCE, 1 },
  { "VSpeedSensor", SOURCE, 1 },
  --{ "SetLimitSwitch", SOURCE, getFieldInfo("sg").id },
  { "HighestLaunchGoal", VALUE, 140, 20, 300 },
  { "BackgroundColor", COLOR, WHITE },
  { "AltitudeIsInFeet", VALUE, 0, 0, 1 },
  { "VSpeedIsInFeet", VALUE, 0, 0, 1 },
  { "ShowRawSensorData", VALUE, 0, 0, 1 }
}

3djc, 2017-04-03:
This in not a bug, the current set limit for widget inputs is set to 5

kilrah, 2017-04-03:
Sounds like an [XY problem](http://xyproblem.info/)... What is it you're actually wanting to do?

agibson2, 2017-04-03:
The 2 options AltitudeIsInFeet and VSpeedIsInFeet are needed because I could not find a way to determine what the user set the units to for the variometer Altitude and VSpd sensors.  I know that lcd.drawChannel() will show the correct ft or m units but I am not just displaying the value I also need to get it using getValue() to do some manipulation of the values.   When I do a getValue() I need to know if the values returned are in meters or feet.  I let the user set those options in the Widget options to match up with what their sensors are set to (feet or meters).  It would be nice to have something like getFieldInfo('VSpd').unit or maybe there is something like that that I just haven't found it?

2 other options AltitudeSensor and VSpeedSensor are there in case someone has multiple sensors and wants to change which sensor the data is pulled in from.  This could be useful to test 2 varios at the same time with different widgets for example.

The background color is obvious.

The ShowRawSensorData is there to set an option to toggle showing variometer drift or the raw sensor data mainly for debugging while on the field.  This should not be used often but it would be great to be able to toggle it on the field if I want to.

I wouldn't want to use precious switch assignments for setting these kinds of options.  It is more elegant to use Widget options as it is more obvious and settings are kept in a consistent location.

I wanted someone to be able to change the inputs for Alt and VSpd but I realized why would someone ever have 2 variometers in a DLG so I removed them to get it down to 5 options.

kilrah, 2017-04-03:
Looks like it indeed isn't possible to get the selected unit - but the "correct" fix is to add a way to get it since it's obviously important. So as I supposed those 2 options are a poorly suited workaround do a different problem, it makes no sense tackling the options in this context.

agibson2, 2017-04-03:
Thanks for the info.  I will go ahead and add a bug report for that.  Is that ok?

kilrah, 2017-04-03:
Yep perfect.

closed 2019-11-15 by stale[bot]