opentx/opentx #4745: Companion Simulator N266,  at Horus settings of Mainscreens are not stored

hrenz, 2017-04-03:
At Companion Simulator N266 
I can set the Hours mainscreens and user interface with theme and widgets 
thats all fine, works fine, but my settings are not stored.

close simualtorscreen, open again, all my settings are back to start

Modell files are stored and reopened

IAt older V2.20 the mainscreen settings are stored, I am shure

Is it only may Problem or a real Issue?

3djc, 2017-04-03:
Simu launched from within companion do not store anything. Are you talking about the standalone simu or companion one?

1Sven, 2017-04-03:
I have the same thing, it is on the standalone simu.

kilrah, 2017-04-03:
Are you sure you're then reading the proper location into companion? Horus standalone simu can raccess data from multiple different locations, what you select is obviously where the changes will be applied, need to read from the right place to see them.

hrenz, 2017-04-04:
I have tested Companion N366
from there started the Simulator the and set the Horus Mainscreens

mpaperno, 2017-04-04:
There are 3 options for Horus data (radio & model settings) when starting the standalone Simulator.
1. A .otx file (or zip archive containing RADIO and MODELS).
1. A folder containing RADIO and MODELS folders.
1. Using RADIO and MODELS in the usual SD card path (this is how an actual radio would start).

So, which way(s) are you using when the issue occurs?  Have you tried using the other methods to see if the problem persists?

kilrah, 2017-04-04:
> I have tested Companion N366
from there started the Simulator the and set the Horus Mainscreens

When launched from Companion the simulator never saves any change, it has been that way forever.

Use the standalone "Firmware simulator" with an appropriate data source if you want changes saved.

closed 2017-04-04 by kilrah

mpaperno, 2017-04-04:
Well, @1Sven did specifically mention the standalone simulator, so I'm still wondering if there's an issue here or not.

kilrah, 2017-04-04:
Ah yeah, missed that... might be some confusion there :\

1Sven, 2017-04-04:
I have changed to "Folder" then it works. I think, it is a issue of me...