opentx/opentx #4746: 2.2 flash extern device not work

frobert0, 2017-04-03:
X9E after update to 2.2, flashing external device as at 2.1.9 with 2.2 does not work
At 2.1.9 comes bars
At 2.2 no bar and after some time error message

With 2.1.9 some receiver flashed now with 2.2 unfortunately does not work

sorry for my English, and translated by google

[image]
[image]

3djc, 2017-04-05:
Just tested it, works fine

closed 2017-04-05 by 3djc

frobert0, 2017-04-05:
At 2.1.9 waits longer for boot
And the option goes as in the picture attachment
[image]

bei 2.2. nur durch Vmain
[image]

3djc, 2017-04-05:
You don't need anything special since frsky receiver accept a wide voltage range. A simple cable with 2 wires crossed. Did flash x6r, x8r and s8. So it definitely works

projectkk2glider, 2017-04-05:
Not really both d8r and x8r have `Operating Voltage Range: 4.0~10V`. X9E has 8 cell battery.

But I looked at the differences between 2.1 and 2.2 code and could not find any - the power on delay is the same in both versions.

1Sven, 2017-04-05:
I use a simple 25 Ohm resistor in the + wire. For the scary users...
Testet with many receivers, vario and the redundacy bus.

schwabe, 2017-04-06:
check that you really use the smartport and not the sbus port