opentx/opentx #4747: LUA: It is not possible to get the units for sensors

agibson2, 2017-04-03:
I have a Widget that needs to do some calculations on the vario 'Alt' and 'VSpd'.  The problem is that I need to know if the user set the sensor to feet or meters to do the calculation properly.  I have worked around this by creating 2 Widget options (AltitudeIsInFeet, VSpeedIsInFeet) so that the user can sync up those to Widget option settings to match what they set the sensors to.  This of course requires the user to make those Widget option changes before the sensor works properly.

It would be great to be able to automatically handle this with LUA code so that the user does not have to tell the Widget how they set up units on the sensor (feet vs meters).  Maybe something like getFieldInfo('Alt').units or something like that.  I don't know if that is the right way to handle it... only an idea that I thought of.

I am aware that lcd.drawChannel() will automatically append m, ft, etc but that is only useful if you want to display the value and doesn't help if you want to get the value through getValue() and manipulate it as you don't know if they set the sensor to ft or m.

lshems, 2017-04-04:
Well, then again, this is already an open issue:

#3334

https://github.com/opentx/opentx/issues/3334

agibson2, 2017-04-05:
Thanks for the work on this 3djc.  I guess I should look at finding out how to build companion and the firmware to test changes like this or is there already automated builds somewhere against all new pull requests for testing?

3djc, 2017-04-05:
First of all it as not been merged to next yet, and when it has, it will be available when a new RC is released

agibson2, 2017-04-05:
I understand that.  I was going to try and help test it before it got merged by building it myself with that cherry picked diff manually merged if I had to.  I guess it isn't worth the effort.  You guys know what you are doing I am sure :).

PR #4750 (2017-04-06): Add unit to lua getFieldInfo
Allow to query a telemetry sensor unit using lua getFieldInfo()

this closes #4747 #3334
closed 2017-04-06 by bsongis