opentx/opentx #4748: Q X7 Switches are Missing in Menus

redbeardfpv, 2017-04-04:
I am new to OpenTX so if this isn't a bug please let me know, but I have looked for days for a solution.  
I am on 2.2.0 N366 and since flashing away from the 2.2 build that ships with the Q X7 I can't get any switches to work, The sticks, pots and trims are available but not the switches. 

On the Inputs or Mixer  screen when I try to select a source only the sticks, pots and trims are available.  If i long hold and scroll down to switches I can assign SA but the switch doesn't function (the value in the graph never changes), and if I scroll away from SA in the source after using the long hold to get there it is not in the list anymore.  

I am not sure what other information to provide but if there is someone available to help troubleshoot please let me know. 

Thanks, 

RedbeardFPV

3djc, 2017-04-04:
I'm not sure of what you do exactly, but It seems to be working here.

In INPUTS screen,, use rot wheel to get to source, push rot wheel, field is blinking, move the switch you want, it is selected

[image]

If I then move the switch you see the vallue change (low, mdedium, up)

[image]

[image]

[image]

kilrah, 2017-04-04:
Check the switches are set to the correct types on the "Hardware" page, you probably wiped them through some improper Companion manipulation.

closed 2017-04-04 by kilrah