opentx/opentx #4753: GPSin Horus in OpenTX not working?

Githubbachelor, 2017-04-04:
Does anybody know if GPS in Horus work in Open Tx ?

agibson2, 2017-04-05:
Not yet.  I don't know if it is on the to-do list or not.  I am not a developer... only a user.  For future, It might be a good idea to ask questions like this on rcgroups so that developers don't get tons of general questions in the bug/issue tracker.  Developers and other in-the-know people on there are pretty good about answering questions there.  Just search for Horus... there is a huge thread on the Horus.

3djc, 2017-04-05:
It does, but yes, this is a bug tracker, not a question board

closed 2017-04-05 by 3djc

agibson2, 2017-04-05:
The GPS in the Horus is not implemented yet that I am aware of.  GPS in models connected to a Rx is implemented.

3djc, 2017-04-05:
Well it is, it might not do yet what you want it to, but it essentially is. Since no details was given on the need, hard to detail the answer tho

agibson2, 2017-04-05:
Sorry for the confusion Githubbachelor.  I thought I was helping but I seem to be causing more confusion.

3djc, Wow. Nice.  I didn't know that it was working at all going by the rcgroups thread.  I only thought it worked with FrOS.  Can you probe it for information for lua scripts?  It would be neat to create a location type lua script to direct someone to the last known location using the Horus GPS to know where to point you.