opentx/opentx #4757: Horus/N366 Can't reverse direction of output

RC-SOAR, 2017-04-06:
A user of one of my sailplane setups found this issue which I can also reproduce on my production Horus (mode 2) running N366

1. Go to Outputs menu
2. Highlight CH4
3. Skip to Direction col
4. Press Enter to change direction.  

At this point, a throttle reverse confirmation dialog appears with two options Enter and Rtn.

5. Press Enter to confirm

Expected: arrow changes direction
Actual: arrow remains unchanged.

I tried altering the Throttle.Src in Model.Setup from Thr to some other control but no difference.

As an aside, I'm puzzled as to why OpenTx thinks that CH4 is associated with the throttle stick as there is no explicit assignment in my setup.

kilrah, 2017-04-06:
Nasty... but can't reproduce, so would you have more details or a model file?

RC-SOAR, 2017-04-06:
Either model should reproduce the issue:
https://www.dropbox.com/s/zwcqakkg20bnagx/esoarplus_212_RO.eepe?dl=0

RC-SOAR, 2017-04-06:
A weird issue... after repeated attempting to alter the CH4 direction the radio spontaneously switches RF mode from internal/internal to internal/int+ext. I get a 'radio antenna defective' audio alert, plus the popup.  I then have to reset the mode in the Model.Setup menu.

kilrah, 2017-04-06:
Mmh this is detected as being a Taranis Plus 2.1 file. Is that normal? 
There will be 2 conversions happening, 2.1->2.2 Taranis Plus then Taranis Plus 2.2 ->Horus 2.2.

I fear the worst...

RC-SOAR, 2017-04-06:
Only this exact config worked fine on a previous build, including changing the channel direction, and it's been test flown without issue. I'm just flashing back to N363...

RC-SOAR, 2017-04-06:
Okay, so I've reflashed and verified that it works fine in N363, I'm able to alter the direction of CH4 with no problem.

kilrah, 2017-04-06:
Opening the file in companion 2.1 with a T+ profile results in no conversion (makes sense) but trying to reverse CH4 also pops up the warning. Enter however swaps it as intended.

Looks like we have 2 bugs:
- Inversion discarded even if enter pressed (both on Taranis+ and Horus)
- Wrong detection of throttle channel (might not even be relevant/possible anymore...)

kilrah, 2017-04-06:
OK, got the inversion issue, I broke it without being aware of it... now on to the throttle channel deteciton

RC-SOAR, 2017-04-06:
Yes I'd always ignored that warning... What's also weird is the spontaneous switching of antenna mode as described above - I was able to repeat it on demand. It seemed to be directly related to my unsuccessful attempts to alter the CH4 direction.

kilrah, 2017-04-06:
the wrong throttle warning is also "normal"... it's obsolete and assumes any channel that has a line with Thr as source is "the throttle channel". That was obviously thought for basic setups and not some with compensations like yours.... and it makes no sense anymore since inputs exist.

Guess we'll jsut throw the whole warning away, impossible to guess what the user is doing. 

Next step, the antenna issue... that's really weird.

kilrah, 2017-04-06:
OK can't reproduce the antenna switch after the fix, so I guess it's been fised at the same time. Makes sense too, the popup confitmation likely casued the switch next time the mode setup page was loaded becasue it was never processed by the outputs page.

RC-SOAR, 2017-04-06:
Thanks for the quick response. Just to be clear, the audio and popup alerts seemed to be triggered after a menu navigation operation, not related to the model.setup menu. Just wondering if I should be concerned about the RF section as I didn't have an external antenna connected.

closed 2017-04-06 by bsongis

kilrah, 2017-04-06:
Yeah it's all correct, trying to invert a channel that had a Thr mix casued the popup and audio warning, confirming it sets a global value that normally shold be processed on next screen refresh but since I deleted the code that I thought was dead it wasn't. 
Now the model setup page uses the same mechanism so next time it was loaded it saw the global var set and interpreted it as the confirmation for antenna setting change.

I doubt your RF section suffered, I played a lot with mine doing measurements and other tests back when the switching was implemented and all that without issue. 
Obviously do a nice good range check anyway before flying again...

RC-SOAR, 2017-04-06:
Okay, and thanks again for super swift resolution :)

PR #4758 (2017-04-06): Check for confirming inversion of throttle channel is no more possibl…
…e since Virtual Inputs got introduced (Fix #4757)
commit 8794f458 (2017-04-06): Check for confirming inversion of throttle channel is no more possible since Virtual Inputs got introduced (Fix #4757) (#4758)