opentx/opentx #4759: Feature to simplify Widget coordinates using lcd.draw...

agibson2, 2017-04-07:
Currently when you draw into a Widget area using something like lcd.drawLine(...), you have to use the Widget area offset from the main screen coordinates (zone.x and zone.y for example).  That is obtained from zone and the create() function.  It would simplify Widget programming if the coordinates 0,0 always referenced the first top left pixel of that specific instance of the Widget instead of needing to add the zone.x and zone.y offsets manually.

projectkk2glider, 2017-05-05:
I agree, this would be more user friendly, but if we do this we affect all existing widget scripts.

agibson2, 2017-05-05:
I am sure it is not ideal but maybe have a feature function that sets the new functionality.  I am not sure how cramped code space is though so maybe having the if conditions for a flag like that is not a good idea.

Something like...
lcd.UseWidgetOffsets(true)

Breaking Widgets with changes is I am sure heavily frowned on as I wouldn't be happy if my Widgets stopped working... then again... there is no official release of 2.2 yet ;).

strgaltdel, 2017-05-08:
maybe one more parameter could be added into corresponding functions.
say "0" would mean "absolute coordinates", "1" would mean "relative" coordinates to widget zone.
If this parameter wouldn't be obligate, downward compatibility of scripts could be realized:

lcd.drawLine(X1, Y1, X1, Y2, PATTERN, FLAGS [,POSITIONING])

3djc, 2017-05-08:
It can be easily done within lua itself to suit programmer preference, and there is much more to widget drawing than offset. And it cannot be for one item drawline for example, it would need to be done for each and every lcd affecting function

strgaltdel, 2017-05-08:
in every widget, some codelines which prevents the script to run in "wrong" / "to small" areas should be obligate.
personally it was a learning effect when i tried to choose a tiny topbar widget the first time and scrolled thru some "normal" sized widgets (-;

but i do also appreciate the "freedom" of using some space beyond intended widget limits in very special cases, where the layout presets doesn't really match my requirements.

btw
" And it cannot be for one item drawline for example"
that's why i wrote " corresponding functions."

imho, in my eyes, the requested feature is not a really an enhancement, but rather a question of convenience

on4mj, 2017-05-08:
IMHO, drawing should be restricted to the widget area.  If everyone starts writing outside their allocated space, there will be overwrites.  On the Taranis, you don't write on other pages, after all.

And I don't see the need to be backward compatible with something which (big quotes here) "doesn't exist yet".  The current status might be release candidates, but people should still expect design changes at this stage.  This won't be the case after the 'RC' disappears from the version string.  
And  a compatibility function, or worse, additional parameters to the draw functions would just add clutter.

Even if what is implemented is only a convenience which computes the origins, it's best done once in the firmware than in a lib that nearly every widget author will write in the end (I already wrote mine).  Better once in C than x times in LUA.

3djc, 2017-05-08:
We do not intend to bring any new/change features to 2.2.0 at this point in time.Something like this (I mean any of the options discussed here) would have to be considered for 2.2.1 anyway

on4mj, 2017-05-08:
Then better never do it at all, I guess.
Adding it later would only bring more work for script authors to support the different firmware versions.

3djc, 2017-05-08:
Not necessarily, you could think of a solution which deal with offset and also would return 0,0 as zone offset, so still work with existing scripts

jfrickmann, 2019-09-22:
This would be brilliant! Now that we may soon get key event in widgets, I am starting to look into how to port my SoarOTX project from Taranis to Horus. And having to go through all of my scripts to add offsets to X and Y everywhere would be a nightmare! 
How are the chances that this will be implemented soon?
As always, thanks for all of your brilliant work!!

raphaelcoeffic, 2019-09-23:
@jfrickmann I can try to have a look in case I find a nice way to specify it. Maybe some option, somewhere.

lshems, 2019-09-23:
Please clarify first on how we can declare widget location and size, and within which restrictions (if any).

I would hate to lose the possibility to use two widgets on top of each other to overcome transaction count errors on drawing too many graph lines or grid lines in one widget.

raphaelcoeffic, 2019-09-23:
Nothing would be lost. The plan is to have the option to use a transparent offset, so that (0,0) is the corner of your zone, and not the corner of the screen. That’s it. It’s not a must to use it or change anything in existing widgets.

lshems, 2019-09-23:
Ok. Sounds good. But will there be a check on drawing withing widget or even within screen? Currently a dangerous sport, drawing on the LCD screen

raphaelcoeffic, 2019-09-23:
I was not planning on improving the clipping, but that right, it would needed...

agibson2, 2020-03-22:
Please keep this going.  I think it is just a valid today as it was when I created the feature request.  Using widget offset for drawing instead of screen offset would be very helpful.

3djc, 2020-03-22:
This is planned for 2.4, where each widget will have its own window space

closed 2020-12-18 by stale[bot]