opentx/opentx #4760: Timer one dissappaers

TerryPts, 2017-04-07:
I have searched and found nothing about this. Bug or my stupidity? When I program timer one it works but when I turn the TX off and back on, timer one disappears (goes back to default). Timer 2 works as it should and both timers work on other models but not the one I need. It does the same thing on the TX and Companion. This is on a Taranis (not plus) running 2.1.9. I'd sent the model file but this won't allow bin files.

kilrah, 2017-04-07:
Please detail exactly what settings you're altering that don't stick.

TerryPts, 2017-04-07:
Hi Andre,

 

Thank you for getting back to me. 

Timer 1                 THs    00:00:00

Name                    flight

Persistent           OFF

Minute call          unchecked

Countdown        Silent

 

Timer 1 goes to OFF and the minute call gets checked when the radio is turned off and then on.

The model file that is causing the problem is attached.

 

Thanks again,

Terry Pitts

TerryPts, 2017-04-07:
To clarify, the x650 works fine the two x750’s are the problem. I copied the model to see if the second copy worked differently… it did not.

kilrah, 2017-04-07:
Attachements aren't supported.

TerryPts, 2017-04-07:
How do I send you a model file?

kilrah, 2017-04-07:
>  Your file can be posted to any third party file host and the link posted here.

E.g. filedropper.com

TerryPts, 2017-04-08:
Hi Andre,

I just got back on my computer. Here is the link to the file…

 

https://www.dropbox.com/home?preview=x750+backup.bin

TerryPts, 2017-04-10:
I’m not in a panic for an answer but want to be sure you got the link I sent on Saturday.

Regards,

Terry

kilrah, 2017-04-11:
Nope, this would only work when logged in, you need to use the "Share" link.

Oldgibber, 2017-05-28:
A link to another forum reporting, I think, the same bug.
[http://www.modelflying.co.uk/forums/postings.asp?th=127532&p=2#2189456](url)