opentx/opentx #4773: Companion 2.2 Duplicate model sometimes launches Model Wizard

elecpower, 2017-04-11:
Post PR #4769 the model is duplicated but then the Model Edit Wizard is launched. Behaviour traced to the Application setting for Action on New Model.
Does not make sense to follow duplication with action based on setting.

Ubuntu 16.04 Taranis X9D+

mpaperno, 2017-04-11:
Good point.  Do you think it should still launch the model editor instead?  Perhaps only if the "action on new model" preference is set to Model Editor?

elecpower, 2017-04-11:
Can see a case to always launch Model Edit as unless making a backup one would expect to make some changes to the duplicate.
Leaning towards take 'action on new model' if not Model Wizard set ie Model Editor or do nothing so it leaves the behaviour to the user

mpaperno, 2017-04-11:
Sounds good to me (if nothing else user probably "should" rename the model anyway).  Now it is so.

commit 350c3a6d (2017-04-11): [Companion] Do not run model wizard on a duplicated/copied model (fixes #4773); Replace existing file extension with .otx when first saving a converted bin/hex/eepe file (fixes #4772); Make sure no other models exist before setting first model as default (fix index 0 always being auto-selected).
closed 2017-04-11 by mpaperno