opentx/opentx #4777: [Companion] Can't open Sky9X settings file w/out it getting converted

mpaperno, 2017-04-11:
Steps to recreate:
1. Set model type to Sky9X.
2. New file.
3. Create a default model.
4. Save and close file.
5. Open the file.
6. Companion thinks the file is for another radio and "converts" it, resulting in warning prompt, a modified file, etc.

Companion seems to think it is a stock 9X/AR9X file:
`[D] OpenTxEepromInterface::loadFromByteArray():198 - "OpenTX for ar9x board / 9X: OK"`

If opening a .bin file, the firmware thinks it's a 9XR-PRO file:
`trying OpenTX for 9XR-PRO import... version 218 variant 0 ok`

bsongis, 2017-04-26:
I just tested and had a SEGFAULT on step 5!

bsongis, 2017-04-26:
Both SEGFAULT and wrong conversion message fixed

commit 3aceed62 (2017-04-26): [Companion] Fixes #4777 (#4860)
closed 2017-04-26 by bsongis