opentx/opentx #4778: Theme- Screensettings are global

1Sven, 2017-04-12:
If in the modelsettings for one model the theme is defined, it’s for every model the same. It’s like a systemconfiguration, it works global. So it is not possible to make for every model a new theme.
May be it is not a really bug, but then it is not the right structure.

kilrah, 2017-04-12:
That's how it's intended. Theme is meant to be for the whole radio UI.

closed 2017-04-12 by kilrah

Enggmaug, 2021-04-29:
"having different looks per model has never been considered"
I would like it to be considered. I'm not asking for you guys to change the entire code structure, so as an alternate method, Is it possible to change the theme with a lua script, when the model is changed ?