opentx/opentx #4780: Companion N368 - Managing Categories

Alex00101, 2017-04-12:
Since I have upgraded to N368 there are various side effects when managing categories. The most serious one is when I read models from radio all models get moved to an unknown category. In addition, the only way to rename the category is to double click on a very particular  spot on its name, menu and toolbar do not work. Also there seem to be an issues when writing categories back to the radio. Windows 10.
Reinstalled fresh on a different PC - same results. 

Thanks

mpaperno, 2017-04-12:
Fixed by 111bcad .

closed 2017-04-12 by mpaperno

Viruswelt, 2017-04-12:
When I create a new category under Companion, the program crashes.
OSX 10.12.5

kilrah, 2017-04-12:
Yep, known and fixed as can be seen above, wait for next version.

Viruswelt, 2017-04-12:
ok !
thx