opentx/opentx #4783: Inconsistent behaivour between Tx and Companion.

rdeanchurch, 2017-04-12:
N368, RC15 Companion on Win10
On the transmitter, deleting a model results in a blank model and the models following stay in the same place.  In Companion, the models are deleted and all following models move up.
I believe the tx method is preferable and would like to see Companion behave the same way as the tx.
In Companion, inserting a model moves all the models following it down.  This option causes model numbers to not reflect receiver numbers if a person uses that as part of his schema.  The tx does not have an insert option.  gain I find the tx performance, preferable.

mpaperno, 2017-04-12:
You can restore the previous behavior (to match radio) by unchecking the option in Companion `Settings -> Application Settings -> Remove empty model slots...`

Also you do not have to insert a model.  You can use an existing empty slot (if you have one), just like before, and paste/drop a model onto there.  Insert is a purely optional feature.

closed 2017-04-12 by mpaperno

rdeanchurch, 2017-04-12:
Thank you, my bad, I missed that checkbox.
I think it is you I owe a giant thank you to.
Companion is very much improved and I really appreciate your efforts.
Especially nice is the new Compare models.

mpaperno, 2017-04-12:
No worries, it's a somewhat significant behavior change and should maybe be highlighted more in the release notes.  I really couldn't decide, perhaps the new behavior should be off by default. 

And thank you.  😺

powercycle, 2017-04-16:
Maybe, since it is new behaviour you should change it to off by default, then move it to default at some future release.