opentx/opentx #4786: Taranis becomes mass storage after 2.2 update

iglogix, 2017-04-13:
After update to OpenTx 2.2 version, when I connect USB to transmitter it immediately becomes mass storage device instead of joystick, as was the case before the update. As a result I cannot use it with simulators etc.

3djc, 2017-04-13:
Masstorage is an option, and still is with 2.2, please make sure you have the right firmware option.

closed 2017-04-13 by 3djc

atheimer, 2018-03-02:
I have run into the same issue.  When I plug in the USB 2 explorer windows pop up.  This never happened prior my update to 2.2.1  Mass Storage does not appear as an option in Companion version 2.2.1 for me using the X9D Radio Type (i have a Taranis Plus).  Here is the screenshot of my build options:

[image]