opentx/opentx #4787: Cannot load download firmware 351

RageFPV, 2017-04-13:
[image]

Companion 2.2 finds the nightly builds, downloads it, but cannot load th efile

RageFPV, 2017-04-13:
when i launch companion 2.2 i get a dll error?!
[image]

3djc, 2017-04-13:
Please describe better the steps you do and the version you are using

mpaperno, 2017-04-13:
Re: the 2nd error (DLL), could you check the dates on the files in your `C:\Program Files (x86)\Intel\iCLS Client` folder and report back?  Specifically date of `ssleay32.dll` and also the size might be useful.

Companion keeps running after the error is dismissed?

kilrah, 2017-04-13:
What "351" are you talking of?

If you're talking of OpenTX 2.2.0 RC1 (N351) you should not download that but use the current RC15, see website homepage.

RageFPV, 2017-04-13:
Sorry, was very tired and put minimal info in last night, full details:

Running up to date windows 10.
From 
C:\Program Files (x86)\Intel\iCLS Client
ssleay32.dll is 297KB and datated 10/3/2014
Companion opens after that error. 

Downloaded companion 2.2 RC15 368 (dont know why I typed 351)
[image]
In application settings I enabled both:
Use OpenTX firmware nightly builds
Use Companion Nightly builds

Ooops, I clicked write backup to radio, rather than write firmware to radio (can be seen in picture above lol). so that is one issue resolved! Problem in chair not in computer!

But the dll thing is annoying and no idea what that function is responsible for an what it may affect down the line.
There is a note next to the downloader for RC15:
"OpenTX Companion 2.2.0 RC15 (N368) - Windows Installer If you get an error about “api-ms-win-crt-runtime-I1-1-0.dll missing” install MS VC++ runtime."
When you visit the link is seems specific to non windows 10 installations, and the error im getting is not referencing a missing "api-ms-win-crt-runtime-I1-1-0.dll"

3djc, 2017-04-13:
Apparently, the error you are seeing is related to Intel Trusted Connect Service Client. If you don't need it , you should remove it, or update it if you do.

kilrah, 2017-04-13:
Yep that looks not to be related to companion, but a generic issue on your PC. The way Windows works it is possible for an outdated lib installed by an app to take precedence over the correct one, seems that's what's happening.

mpaperno, 2017-04-13:
Not sure it's so simple guys.  That Intel thing seems pretty standard, and also it is based on OpenSSL (that's what the DLLS are from).  I have it on my Win7 here, but mine is not in the PATH and hence I don't see any warnings. But I do see warnings from Qt when checking for updates about SSL not working.   MinGW comes with its own ssleay/etc DLLs which I copy into the OTx app folder so there are no conflicts and no SSL warnings.  But I'm not clear which OpenSSL DLLs should be included with the MSVC version (or where to get them).

I would suggest making sure the Intel iCLS folder is removed from the PATH env. variable, but that is likely to break any other apps which depend on SSL.  Likewise I wouldn't go uninstalling Intel iCLS unless you're very sure it's not used by anything.

3djc, 2017-04-14:
Closing this as it seems rather more a user PC issue than an Otx bug

closed 2017-04-14 by 3djc

kilrah, 2017-04-14:
We might still have a packaging problem, if we need ssleay32.dll then it should probably be distributed...

I just had a look and it's weird. When I check for updates companion indeed loads ssleay32.dll and a couple of other OpenSSL libs that it finds in the PyQT4 install dir (which is in the PATH). If I remove it from the PATH it does not load them (i.e. they aren't found anywhere else) but the update check works anyway.

3djc, 2017-04-14:
Well we could if we can ensure it's taken prior to the faulty one. I have a total of over 30 on my PC, seems to be quite messy  to guess which one will be used

kilrah, 2017-04-14:
Normally if there is one in the same folder as the executable it takes precedence over everything, then the system32 folder, then the rest in the order of the PATH environment var.

But it seems it isn't actually needed - best would be to understand what tries to load it before falling back to another method if it can't and remove that.

mpaperno, 2017-04-14:
Unfortunately it seems we need to either build our own Qt with SSL disabled entirely, or ship OpenSSL DLLs with the app, even though no SSL is actually being used.  http://stackoverflow.com/questions/26361145/qsslsocket-error-when-ssl-is-not-used