opentx/opentx #4789: ON363- Special Function issues

clivemartin, 2017-04-13:
With 2.2 ON363 special functions not working for the assignment of Adjust GV's, Volume, Play Value and Vario.

schwabe, 2017-04-13:
A bit more context please, which radio? Companion or Radio? What exactly is not working.

clivemartin, 2017-04-13:
Problem is with companion, both Horus and 9XD radio profiles.  Start new model, go through Wizard basic set up, in mix panel set up a GV for weight (GV1) on any channel, go to Special functions to assign a source for adjusting GV1  eg S1, that source cannot be set, dropdown menu goes grey, also same effect with trying to set Volume, Play Value and Vario.

Also have noticed same faults when trying to edit downloaded model files/setups from transmitters.

mpaperno, 2017-04-13:
Pretty sure this is a dupe of #4779 and fixed now.

closed 2017-04-13 by mpaperno