opentx/opentx #4795: RC16 - Q X7

MIKE-DAILEY, 2017-04-15:
On Q X7 cannot clear a line in Special Functions.  Click clear and the line is still there.  Works OK on the Horus though.

3djc, 2017-04-15:
What version are you using ? Works here, both on simu and real radio

kilrah, 2017-04-15:
Yup, can't reproduce at all.

MIKE-DAILEY, 2017-04-15:
I think we had this problem way back on one of the early builds.  Can't remember if there was a fix or we had to rebuild all the Special Functions.  If I start with a new model the clear works.  Note in the images below from an exiting model there are extra stuff in the pop-up window. 

View of the Special Functions Screen: 

[image]

MIKE-DAILEY, 2017-04-15:
Long Press:

[image]

MIKE-DAILEY, 2017-04-15:
Select an click clear and note the other stuff on the screen:

[image]

MIKE-DAILEY, 2017-04-15:
After click pop-up goes away and line is not cleared:

[image]

kilrah, 2017-04-15:
Are you sure you're actually running RC16?!

MIKE-DAILEY, 2017-04-15:
Shoot!  Forgot that screen. 

[image]

3djc, 2017-04-15:
Could you post your model? You have a combined menu that I don't understand

MIKE-DAILEY, 2017-04-15:
Sure. 

[QX7Models.zip](https://github.com/opentx/opentx/files/923596/QX7Models.zip)

kilrah, 2017-04-15:
Aha, it only breaks when you scroll down, i.e. anytime the first line isn't visible on screen.

MIKE-DAILEY, 2017-04-15:
So I can just wait until there is a fix and not need to rebuild my Special Functions?

kilrah, 2017-04-15:
Rebuild would change nothing at all

MIKE-DAILEY, 2017-04-15:
Oh! Of course not.

PR #4802 (2017-04-16): 3djc/fix #4795
This fixes #4795
commit 1d4ec6f7 (2017-04-16): 3djc/fix #4795 (#4802)

* Prevent popup in undue places

* Working fix ?
closed 2017-04-16 by kilrah