opentx/opentx #4796: Companion Simulator to save window position (debug and tele if rearranged) on OS X

Red-Devil74, 2017-04-15:
If I start up Companion on OS X (10.12.4), open an OTX file and simulate that file, I want the debug window and telemetry window as per attached, so I rearrange this (windows "docked", to the left and on top of eachother. If I close companion (or if it crashes) and reopen Companion, the window position are back to default (undocked, and floating arround) and I have to rearrange this again. I think I saw that on WIN machines the positions are stored in between session. Don't know if this is a feature request or a bug...

[image]

3djc, 2017-04-15:
It does saves on windows and Linux Debian (but cannot say for Mac OS)

Red-Devil74, 2017-04-15:
Haha! But I can! So that would make this a bug then...

mpaperno, 2017-04-16:
Can't recreate here (OSX Sierra), and yes window positions/size should be saved on all OSs.  It only saves when Simulator is properly shut down (not crashed).  It also saves those per radio profile, so if you switch profiles you'll get a new arrangement.  

How do you exit Simulator when launched from Companion?   Also does this happen if you use the standalone Simulator directly?

Red-Devil74, 2017-04-16:
I think I am not doing strange things... see: https://www.youtube.com/watch?v=6kiYOaFtL1U

mpaperno, 2017-04-16:
Thanks for the vid.  I thought maybe it was an issue with full-screen, but that also saves normally on my Sierra (next time it reopens full screen as well).   But just to double-check, does the problem still exist if you _don't_ switch to full-screen/maximized mode before closing?

Do other Simulator settings get saved between uses, for example all switch/pot/slider positions?  Or eg. Word Wrap setting in Debug window.

Red-Devil74, 2017-04-16:
Just checked. 
-Maximised or Not maximised doesn't make a difference...
- switch and pot positions are saved between simulator sessions (so quiting and restarting simu WITHOUT exiting Companion). When I DO exit companion, all switch positions are at default
-For Wordwrap is is the same (saves between simu sessions, not between companion sessions)

Also, maybe not relevant as I do not know if it has been developed but the option to reload de lua environment doesn't work. That is to say, If I have a running simu with a LUA widget in it and I change something in a text editor, press save and then try to reload the LUA, no changes are seen in the widget...

kilrah, 2017-04-16:
Preferences are in `~/Library/com.opentx.Companion 2.2.plist`, you might want to delete/move them and try to start fresh. 
Maybe also run as disk check on your mac and/or a "fix permissions".

mpaperno, 2017-04-16:
Sounds like something is overwriting your saved profile settings when you exit Companion (or, the settings are never being properly saved to permanent storage in the first place and are only kept in RAM).  Please check Companion Settings -> Radio Profile tab, make some change(s) there, and see if it persists between Companion uses.  I'm guessing not.

Reloading Horus Widgets is a known issue.

mpaperno, 2017-04-16:
On mine the settings file is in `~/Library/Preferences/com.opentx.Companion 2.2.plist`.  Unfortunately it's a binary format (unlike Linux/Win where it's easier to see what the actual saved settings are).

kilrah, 2017-04-16:
Really? Same place here, but it's the usual plist XML format...

mpaperno, 2017-04-16:
Yep... At least TextEdit can't read it (shows binary junk), or any other .plist.  Tried opening with XCode but that crashed  :).  I do remember editing .plist files at some point...   Do you have Sierra?

Ah... https://apple.stackexchange.com/questions/101719/trouble-opening-plist-files-in-text-editor

Red-Devil74, 2017-04-16:
Ok, deleted the pref file and started Companion. Made new profile for Horus, opened OTX file, Isseu with incorrect switch warning is still here. Quit programme, opened programme again, changed a bit in the Radio profile setting and restarted companion. The change IS still there so this is saved. Ran "first aid" on ssd. no issues there. ("repair disk permissions" has been removed from Sierra...I mis Steve Jobs dearly)
If you have Xcode installed, the plist files open up nicely in a programme 
[image]

Red-Devil74, 2017-04-16:
ah,.. cross post regarding Xcode...

Red-Devil74, 2017-04-16:
checked the .plist file. there are 2 entries regarding windows:
- mainWindowState
- mainWindowGeometry
both have some hex data value which is the same as chinees to me... No help from this

kilrah, 2017-04-16:
Should be in the profile's simulatorOptions. I've got a nice array in a binary reg value on Win, but it seems on Mac it gets encoded as a string and craps out.
```
<key>Profiles.profile0.simulatorOptions</key>
<string>@Variant(</string>
```

Red-Devil74, 2017-04-16:
I have the same...
[image]

mpaperno, 2017-04-16:
OK, seems we've tracked this down to an older version of Qt on the OTx Mac build server (5.7.0 vs. 5.7.1).  Also affects the switch warnings issue (#4803).

Ref: 
https://bugreports.qt.io/browse/QTBUG-56124
https://codereview.qt-project.org/#/c/174356/

Red-Devil74, 2017-04-17:
@mpaperno Amazing how you tracked that down! Not really clear to me what this means, though. Will this be corrected in a future build on the server side or is the OTX code adjusted?

kilrah, 2017-04-17:
Next build will have it fixed thanks to the QT version having been upgraded on the build server.

closed 2017-04-25 by kilrah

Red-Devil74, 2017-04-27:
@mpaperno @kilrah @3djc  I don't know if closed issues are monitored, but just wanted to confirm that the new release indeed solves my issues with the sim! Thanx Gents! (or at least, I assume you are Gentlemen...  No way to check...)

3djc, 2017-04-27:
Glad to hear it :)