opentx/opentx #4803: Wrong switch position warning Simulator at first run (Horus, N369, Mac OS X 10.12.4)

Red-Devil74, 2017-04-16:
1. -Open companion(N369) on Mac.
2. -Open .otx file (imported from Horus with N369 firmware).
3. -Click Simulate Radio
The screen shows me the "Controle warning" with S1 and S2 at wrong position (see attached)
[image]
I know this to be wrong as more position warning are set and wrong at boot.
4. -Close the simulator
5. -Open the simulator again
The screen now shows the correct warnings (see attached) SA, SB, SE, S1, S2
[image]
[170415.otx.zip](https://github.com/opentx/opentx/files/924238/170415.otx.zip)

mpaperno, 2017-04-16:
Unfortunately can't recreate this either (OSX Sierra).  We'll keep trying, and thanks for posting .otx file.  Can you tell us more about your Mac?

Red-Devil74, 2017-04-16:
It's an MBP
[image]
rather old, but with a samsung SSD in it... Recently I tried to take a stab at compiling and therefor I installed a lot of shit via terminal but it is way out of my league so I stopped... I am having serious other issues with lua scripting which makes me doubt if it is not my (machine) that is acting up... I don't know how to remove all OTX stuff and start from scratch... As far as I can see, nothing is installed in other locations than "applications"... I there something in library that I can delete, or in preferences?

Red-Devil74, 2017-04-16:
What info are you looking for,.... I call it " Johny",... the SSD is called " Max Drive" and the old HDD is called "Big Mac",... they don't listen to me any more than my kids do... ;)

mpaperno, 2017-04-17:
Underlying cause is duplicate of #4796, so will close this.

(Sorry, storage and offspring issues are out of scope here :-D  )

closed 2017-04-17 by mpaperno