opentx/opentx #4804: Companion 2.2 sam-ba

josephdouce, 2017-04-16:
error could not find sam-ba.exe 

http://i.imgur.com/Vwa3TvO.png

change default path to point to C:/Program Files (x86)/OpenTX/Companion 2.2/dfu-util.exe fixes the issue

schwabe, 2017-04-16:
sam-ba and dfu-util different tools for different chips. Pointing sma-ba to dfu-utils sounds wrong. This might be a typical [XY problem](https://meta.stackexchange.com/questions/66377/what-is-the-xy-problem)

josephdouce, 2017-04-17:
I don't think the issue is that sam-ba is missing, I think the issue is it is looking for it even though it doesn't actually need it, since its a fallback, pointing it to any file that actually exists fixes the issue. It is a 2.2 is as 2.1 works fine with no sam-ba.

josephdouce, 2017-04-17:
Just tested it and pointed it to notepad++.exe and everything works fine I think it's trying to call the .exe file as well as using the bootloader. So if you point to a file that doesn't exist you get an exception. Just tested all functions with it pointing to and from a random.exe I can load and upload models and radio settings and backup firmware.

benoit35690, 2017-05-17:
I do reproduce the bug with a 9XR-PRO.
In "Settings" | "configure communications"
For the Taranis :
I have "DFU" configuration with this location
C:/Program Files (x86)/OpenTX/Companion 2.2/dfu-util.exe

For the 9xR-Pro :
I have "Sam-ba" configuration with this location
C:/Program Files (x86)/OpenTX/Companion 2.2/sam-ba.exe

Does the 9XR-PRO requires Sam-ba ?

kilrah, 2017-05-17:
That is correct. SAM-BA is what's used to flash a 9XR-Pro if the processor is blank/erased.
When trying to flash companion looks for the radio's virtual disk drive. If it's found it writes firmware to it, if not it launches SAM-BA instead.

awshima, 2017-07-29:
Same here, everything works fine with 2.1.9, but it keeps asking for sam-ba.exe with version 2.2.0
The internal memory and SD are acessible via Windows Explorer as always, and the correct path is configured in Companion's settings.
I could notice one difference on these settings from version 2.1.9, the SD Structure path now shows a forward slash for the path (Linux style) instead of a reverse slash (Windows style).
In my computer the SD is in drive H so in version 2.1.9 I can see H:\ in this setting and in version 2.2.0 it is H:/
I don't know if it is relevant, but it may be something to look at.

jimpriest, 2018-03-17:
Still an issue.  On Linux (Ubuntu) this fails.  Can't update radio.

kilrah, 2018-03-18:
Becasue it wasn't closed?

jimpriest, 2018-03-18:
Sorry - was looking at 4925 and got them mixed up...  I reverted back to 2.1.9 and things worked fine.

blairfancy, 2019-01-23:
I m facing with same issue. after diging into problem what i see is that files are properly saved in /tmp/xxxx-emp.bin but for any reason companion are not able to read file automaticaly.
But if you open the file with menu file/open it work properly.

DivideZAR, 2019-04-11:
Is there any solution to this problem ? 
Does it mean that even though they say they support th9x they really don't ?

JAZ-013, 2022-05-25:
Just installed Companion 2.3.15 (latest as of now) and this is still an issue.  Is there a fix?  Can I grab the file from somewhere and copy it into the folder?