opentx/opentx #4805: Channel an Mixer Monitor

Gewe99, 2017-04-16:
Hi to all...
there is a tiny problem with the channel an mixer Monitor.
The square brackets which show the limit of a servo are wrong if the servo is inverted.
You can see that clearly if e.g. the left limit is minus -25% and the right one +75%.
The bargraph goes, when servo is inverted, into the correct direction but exceeds the limits (square brackets). (or doesn't reach it..) The brackets have to be inverted as well.
Have a look at channel 18
[image]
Regard 
Dirk

3djc, 2017-04-16:
Please post your model otx

Gewe99, 2017-04-16:
It is the f3jGvrTsw...

Regards Dirk

kilrah, 2017-04-16:
Email attachments aren't supported. You need to upload to Dropbox or equivalent and psot a link.

Gewe99, 2017-04-16:
https://www.dropbox.com/sh/4ord7enzysega8b/AACUhzWVZgEpLYx5apx37eV7a?dl=0

Hope that works. 
I am not thar familiar with Dropbox....

Von Samsung-Tablet gesendet

PR #4813 (2017-04-18): Invert outputs limits when output is reversed
This fixes #4805
closed 2017-04-18 by bsongis