opentx/opentx #4815: Taranis 2.2 does not connect as gamepad anymore

TempleClause, 2017-04-17:
After updating to 2.2, the USB gamepad thingy does not work anymore. 
I power on my Taranis, clear all the warnings and connect the USB cable. But now it gets connected like it's in the bootloader mode (SD card is accessible and I can read and write with the companion app). 

After I unplug the USB cable the Taranis restarts. 

Any ideas?

I'm on Windows 10 by the way.

kilrah, 2017-04-17:
You selected the "massstorage" option. Don't.

closed 2017-04-17 by kilrah

TempleClause, 2017-04-17:
Where is that option?

kilrah, 2017-04-17:
In the radio profile in compamnion preferences, that you set before downloading firmware.

TempleClause, 2017-04-17:
gotcha! Thanks for instant help! No idea why oscarliang recommends having this option selected...

kilrah, 2017-04-17:
No good reason for recommending it indeed...

kurtc021914, 2018-01-18:
On companion 2.2.1 i don't have the massstorage option... :(

kilrah, 2018-01-19:
Read the 2.2.1 release post on the website, it's explained how it works now.

kurtc021914, 2018-01-19:
@kilrah you talk about that ? : 
"Added USB Mode to radio menu setup screen (Ask, Joystick, SDCard)" ?
because it doen't help me. If you could tell me more i would appreciate. Thx