opentx/opentx #4816: X7 sport update external issues

kilrah, 2017-04-17:
A few people have reported sport update of external accessories not to work.
Found that `sportUpdatePowerOn()` calls `SPORT_UPDATE_POWER_ON();`, which for X7 is defined as...  `sportUpdatePowerOn()`.

Guess we've got a loop there.

At the same time we should set the power to the bottom connector to always be on in normal use since people would like to connect things there.

bsongis, 2017-04-24:
I don't see the same code:

```
void sportUpdatePowerOn()
{
  if (IS_PCBREV_40())
    GPIO_SetBits(SPORT_UPDATE_PWR_GPIO, SPORT_UPDATE_PWR_GPIO_PIN);
  else
    EXTERNAL_MODULE_ON();
}
```

bsongis, 2017-04-24:
Who can give a test at this function? I don't have this connector on my prototype

3djc, 2017-04-24:
First batch, so no connector either

bsongis, 2017-04-24:
I will ask to FrSky to do another test before we can close this issue

bsongis, 2017-04-25:
Test done by FrSky today

closed 2017-04-25 by bsongis