opentx/opentx #4824: Gitbook 2.2 manual: Amber sounds are for other transmitters also

RealTadango, 2017-04-19:
On this page: https://opentx.gitbooks.io/manual-for-opentx-2-2/content/advanced/add-ons.html it looks like the amber sound pack is Taranis only. It works on other (all??) OpenTX transmitters so a general section mentioning the sounds would be clearer.

3djc, 2017-04-19:
- it doesn't specify that this is english only
- since the amber pack download does not specify compatible sdcard version number, unsure it should be mentioned  there
- the doc as its own repository where issue should be created : https://github.com/opentx/opentx-taranis-manual

RealTadango, 2017-04-19:
Ok, so the repo for the taranis and 2.2 manual is the same? I will change the issue to the manual repo

3djc, 2017-04-19:
Yes, there is a 2.2 branch there :)

closed 2017-04-19 by RealTadango