opentx/opentx #4825: Companion2.20N369 Behaviour of Delete in Special Functions

bryan39-2, 2017-04-19:
Running the latest DEB version on Linux Mint 18.x (all up to date).
Right clicking on a line in Special Functions and selecting Delete garbles the listings. See the before and after attachments when I deleted SF6
[image]
[image]
If I close the window and re-open it, all appears to be well. I have not tried saving the garbled display to see what happens.

Oldgibber, 2017-04-19:
Copy and paste does the same. (Windows 10)

closed 2017-04-24 by mpaperno

mpaperno, 2017-04-24:
Until next release, workaround is to close and re-open the model editor.  The actual actions are not being modified, the problem is display-only.

commit f9b7f567 (2017-04-24): [Companion] Fix Special/Global Functions action selector being reset for all lines when one line is deleted/cut/pasted (closes #4825).