opentx/opentx #4827: Companion2.20N369 Simulator crash with a multidimensional array

austrianglider, 2017-04-19:
When using a multidimensional array in a lua-script, the simulator crashes when the script is called. 
When the script is called at TX Horus, the Horus goes into the emergency mode.

Q: Is the array to big? If I restrict the array to 2 value series, then the script doesn't crash.

Example Array:
local freqLookup = {}
    freqLookup[1] = { 5865, 5845, 5825, 5805, 5785, 5765, 5745, 5725 }
    freqLookup[2] = { 5733, 5752, 5771, 5790, 5809, 5828, 5847, 5866 }
    freqLookup[3] = { 5705, 5685, 5665, 5645, 5885, 5905, 5925, 5945 }
    freqLookup[4] = { 5740, 5760, 5780, 5800, 5820, 5840, 5860, 5880 }
    freqLookup[5] = { 5658, 5695, 5732, 5769, 5806, 5843, 5880, 5917 }

projectkk2glider, 2017-04-19:
What kind of script? Widget or stand-alone or something else?

kilrah, 2017-04-19:
What are you doing with this array?
Could be unrelated, say creating the array works fine but then the length causes you try to display it out of screen bounds...

I.e. might want to post the script.

3djc, 2017-04-19:
As a side comment, you are initializing the array, but not the rows : https://www.lua.org/pil/11.2.html

austrianglider, 2017-04-19:
It's a stand-alone script.
With the script you can make settings on the VTX Tramp HV.
After selecting the band and the channel, I would like to display the corresponding frequency.

[image]

The affected code starts at line 305.
[KissTest.txt](https://github.com/opentx/opentx/files/935610/KissTest.txt)

projectkk2glider, 2017-04-19:
This runs OK in simu:

```lua
local freq = {
    { 5865, 5845, 5825, 5805, 5785, 5765, 5745, 5725 }, -- Boscam A
    { 5733, 5752, 5771, 5790, 5809, 5828, 5847, 5866 }, -- Boscam B
    { 5705, 5685, 5665, 5645, 5885, 5905, 5925, 5945 }, -- Boscam E
    { 5740, 5760, 5780, 5800, 5820, 5840, 5860, 5880 }, -- FatShark
    { 5658, 5695, 5732, 5769, 5806, 5843, 5880, 5917 }, -- RaceBand
}

local function run(event)
  print("test")
  return 1
end
return { run=run }
```

So the problem must be somewhere else.

projectkk2glider, 2017-04-19:
The whole posted script does indeed crash simu. I will investigate...

projectkk2glider, 2017-04-19:
Looks like Lua has a hard limit on a number of local varibles https://github.com/tessel/t1-runtime/issues/163#issuecomment-55960052 :

>Hmm, this is actually a limitation of the Lua interpreter, and must be less than 250. It won't be easy to increase, either: local variables are mapped to Lua VM "registers", which are stack indexes within the current function frame. These must fit in a single byte because they are encoded in most of the Lua opcodes as 8-bit fields.

>The default 200 allows some space for temporary registers, needed to compile expressions and function calls. There is also a limit of 60 upvalues - variables captured in the function closure from the enclosing function. I'm not sure why this one is so small.

>Note that the 200 locals limit applies only to active locals within one function - since Lua has proper lexical scoping, a local used inside a loop or 'do' block only exists during that block. Outside the block, the register is freed up for a different local (e.g. in a following loop or block.)

>**Also, the top level of a Lua source file is compiled as a function body, so the same limit applies to locals at the file level. This is usually where the limit is hit ;) Named local functions are also included in the total.**

projectkk2glider, 2017-04-19:
Ha ha, @bsongis disabled this check https://github.com/opentx/opentx/commit/f94b4c743d6720641f985790d71d99d976b831c9#diff-be907d08fbafc22e254d337202fef3ceR289, in this issue the `setnodevector()` gets called with size >255

projectkk2glider, 2017-04-19:
Maybe that is not a problem since max size is much bigger. the issue happens when  `luaH_resize()` is called with `nhsize = 256` (and current size is 255).

projectkk2glider, 2017-04-19:
The commit f94b4c743d6720641f985790d71d99d976b831c9 did this. The size of hash is now stored as straight number (8 bit) while it was a number of bits before. So any hash bigger than 255 can't fit. @bsongis any idea how to fix it?

bsongis, 2017-04-20:
I can't remember anything of this commit!

bsongis, 2017-04-24:
@projectkk2glider will you do a PR with your change? Or is there any problem hidden behind my original change and I cannot remember?

projectkk2glider, 2017-04-24:
I need to study this change a little more to see what is behind the scenes. Then I will do a PR.

PR #4855 (2017-04-26): Projectkk2glider/issue 4827
Closes #4827 

This reverts part of commit f94b4c7 and restores back the original table hash size calculation. The code before this allowed only 265 hash nodes.

The code now potentially uses more RAM (because it allocates node array in bigger steps), but at the same time there are a lot less node allocations and copying of existing nodes in the new array. Previously each node number increase caused the whole malloc/copy/free process, now it only happens when size overflows a power of two (ie 1, 2, 4, 8, 16, 32, etc,...). So there are a lot less calls to the `rehash [... cut]
closed 2017-04-26 by bsongis