opentx/opentx #4831: Czech speech audio issue 2.2N369

janekx, 2017-04-20:
There is issue in 2.2N 364-369. Selected czech sound pack not reproduce the values well. 
the value 10.7V should say "ten point seven volts" 
we got only  "ten seven"

English audio works well. Tested on X7, X9E 

On SD we have sthe sound files well and inclide point, volts ....

bsongis, 2017-04-24:
ten => 10.wav
point = CELA => 114.wav
seven => 7.wav
volts = 118 + (unit-1)*4)+3  **with unit = volts = 1** => 121.wav

bsongis, 2017-04-24:
How were the TTS CZ created? Are there 4 prompts per unit in the TTS CZ pack?

3djc, 2017-04-24:
Yes there are 4 :https://github.com/opentx/opentx/blob/next/radio/util/tts_cz.py

bsongis, 2017-04-24:
but unless I am mistaken we play prompt numbers, this is plain wrong!

mhotar, 2017-04-24:
I have not investigated the "point" problem, it might be the pack problem. But you are right it plays number prompts instead of names eg. volt4.wav

bsongis, 2017-04-24:
First proposal for a naïve no CZ speaker fix:
```
PLAY_NUMBER(qr.quot, 0, ZENSKY);
CZ_PUSH_UNIT_PROMPT(qr.quot, CZ_PROMPT_CELA);
PLAY_NUMBER(qr.rem, 0, ZENSKY);
CZ_PUSH_UNIT_PROMPT(qr.quot, unit);
return;
```

commit f10bad7a (2017-04-26): Proposal to fix #4831 (CZ TTS problem with PREC1 and units) (#4856)

* Proposal to fix #4831

* Next fix for #4831

* Initiate some CZ docs

* Some more structure, still a workframe

* My attempt to help with #4831

* Update tts_doc_cz.md

* copy/paste error

* Update tts_doc_cz.md

* missed one number/qr.quot

* Cosmetics

* Consistency in parameters order

* Nearly complete doc

* Cosmetics to facilitate coding for non CZ devs

* Fix integer+unit, cosmetics

* Update tts_doc Radians and Mililiters are Male gender

* Radian and Millimeter are Male (so i've been told)

* Cosmetics

* Invers pu [... cut]
closed 2017-04-26 by bsongis