opentx/opentx #4832: Czech display issue 2.2.0N369

xrom4n, 2017-04-20:
The display contains errors. Page 1/9 radio settings - the text "center transition" and "mode" exceeds the edge of the display, the text "introductory logo" is written without a space. Page 5/9 version - lower texts go beyond the edge of the screen.

xrom4n, 2017-04-20:
Taranis X7

3djc, 2017-04-21:
For screen 5/9 I can do that (not sure if the means anything at all:
[image]

For 1/9 I'll need screenshot and alternative traductions

xrom4n, 2017-04-21:
It looks good. "zálohuje" could replace "Záloha"
I'm sending the current version of my version on the radio.

[image]

closed 2017-04-24 by bsongis