opentx/opentx #4836: Screen OFF brightness can't go below 5

Red-Devil74, 2017-04-21:
When setting the OFF-Screen brightness (OTX N369) the minimum setting is 5. apperently in previous version(s) it could be switched off completely.
Request to enable OFF-screenbrightness to 0 (zero/nill/null) so to actualy switch the screen off completely.
[image]

kilrah, 2017-04-22:
Intentional.

closed 2017-04-22 by kilrah

Red-Devil74, 2017-04-22:
Thought so much. Thanx. Do you have any prediction on power consumption when screen is at 5%? That is the only reason "someone" would like to have it on 0%, to reduce power. I would assume it can be neglected?

3djc, 2017-04-22:
It does consume more power than at 0% but it's the only way to get out of a stupid settings, so not a choice really

Red-Devil74, 2017-04-22:
Could it be a feature enhancement to check wether Mode is not set to OFF and On brightness is above 65% and THEN allow 0% OFF Brightness?

kilrah, 2017-04-22:
Consumption at 5% is completely negligible.