opentx/opentx #4837: Crash when syncronizing SD card

Freddan-67, 2017-04-22:
First time use on a new computer.
Go to settings->settings and set up a writable directory (in my case %userprofile\opentx\sd-backup\
Go to file->syncronize SD
left-click outside of the checkbox called "show details" 
The application crash.
[image]
[image]

projectkk2glider, 2017-05-05:
@Freddan-67  Which version of Companion?

Freddan-67, 2017-05-06:
As can be seen on top of the 3:rd image I am using 2.1.9 (windows 10), Lenovo X1 Carbon (2016) with Intel USB 3 host.

closed 2017-05-08 by bsongis