opentx/opentx #4838: QX7: Enable DATA port power by default so that accessories can be used/powered by it

kilrah, 2017-04-22:


kilrah, 2019-01-06:
Even better, a choice on hardware page for S-Port power Auto/ON

Tiefflieger68, 2019-09-02:
is it still in scope ?

KnuckleUpFPV, 2019-09-19:
Is it even possible to use the bottom s.port pin while a crossfire is installed in the module bay? Trying to get crsf mirror over that pin, because the crossfire bluetooth is unreliable.

kilrah, 2019-09-19:
All sport pins are hardwired together so with CRSF you get everything on there (both uplink and downlink, need to filter yourself if needed).

KnuckleUpFPV, 2019-09-19:
Im not getting anything from the bottom s.port pin. Its like its dead when crsf is in use. Thanks though.

pulquero, 2019-12-14:
In case anyone finds it useful, I can confirm that it is possible to read smart port telemetry from the bottom port on the Q X7 (opentx 2.3.1) using an Frsky STK usb dongle.

Here is some python code:
```
import serial

def unpack_8(data):
    return data[0]

def pack_8(data):
    return [int(data & 0xFF)]

def unpack_16(data):
    result = data[1]
    result = (result << 8) | data[0]
    return result

def pack_16(data):
    part_a = (data >> 8)
    part_b = (data & 0xFF)
    return [int(part_b), int(part_a)]

def unpack_32(data):
    result = data[3]
    result = (result << 8) | data[2]
    result = (result << 8) | data[1]
    result = (result << 8) | data[0]
    return result

def pack_32(data):
    part_a = (data >> 24)
    part_b = (data >> 16) & 0xFF
    part_c = (data >> 8) & 0xFF
    part_d = (data & 0xFF)
    return [int(part_d), int(part_c), int(part_b), int(part_a)]

ser = serial.Serial(port='COM11', baudrate=57600, timeout=0.1)
while True:
    header = unpack_8(bytearray(ser.read(1)))
    if header == 0x7e:
        payload = bytearray(ser.read(1))
        physical_id = unpack_8(payload)
    else:
        payload = bytearray(ser.read(7))
        sensor_id = unpack_16(payload[0:2])
        value = unpack_32(payload[2:6])
        print '{0:02x} Sensor {1:04x} {2}'.format(header, sensor_id, value)
```

kilrah, 2020-06-11:
Done already

closed 2020-06-11 by kilrah