opentx/opentx #4839: Number of Widgets Crash Companion Latest release

aviat40, 2017-04-22:
Around round 10 widgets crash the companion
Thierry71

projectkk2glider, 2017-04-22:
Could be the same reason as #4827

bsongis, 2017-04-24:
Any example of such widget? You don't provide much information!

aviat40, 2017-04-28:
I will try to be more precise. To understand the Widget, I create around 10 folders with all widgets i can found on internet with the Horus. I use the Simulator openTX lastest version using Folder option to simulate. 
After selected, when you want to start the simulator, the App crash. OS W10

kilrah, 2017-04-28:
Please zip your WIDGETS folder and post it.

aviat40, 2017-04-29:
[Widgets.zip](https://github.com/opentx/opentx/files/965925/Widgets.zip)

I hope it will help you

kilrah, 2017-04-29:
Nothing to do with the number, it's the "BrainFlightMode" widget that causes the crash.

PR #4876 (2017-04-29): Fixes #4839: path buffer overflow if LUA widget or theme directory na…
…me was too long (max length of directory name is 13 characters)

Closes #4839
commit 582546f8 (2017-04-29): Fixes #4839: path buffer overflow if LUA widget or theme directory name was too long (max length of directory name is 13 characters) (#4876)
closed 2017-04-29 by kilrah