opentx/opentx #4840: [Companion n396] SF Conversion from older releases & between tx profiles "buggy"

strgaltdel, 2017-04-22:
Hi,
just updated my X9e from n366 to n369 and wanted to write a model into my horus.

Following Problem:

in n366 a SF is defined to control speaker volume:
By flashing n369 on X9e everything seems to be fine within TX model configuration.
[image]

After that i loaded the "pre saved" n366 otx file into companion n369, still "X9e" in profile settings.
There i recognized, that the SF config changed from input "RS" into "LS", which didn't occur by firmware conversion.
[image]

Without changing this config i switched to my horus profile and could see, that the input ended up in the "rubber stick" / x level:
[image]

Didn't search for more changes, but would expect them
Could find a similar issue
Don't know if it has something to do with my X9e setting: this tx is "expanded" with 2 more switches and the 6Pos pod

regards
Udo

3djc, 2017-04-22:
X9E had (and still has until next nightly) invalid handling of two pos additional switches. If you had one or more of those switches, they would very likely be wrong (ie your display position is middle and up/down instead of the correct up/down (no middle)).

In addition, it is nealy impossible to go from X9E (specially expended) to anything else, since X9E has more inputs than any other Otx radio

strgaltdel, 2017-04-22:
Extension switches are 3-Pos types, so it doesn't perfectly match your failure criteria !
Never saw this kind of "Input shift" by any other Companion upgrade
[image]

I'm fully aware that there is no option to copy a model "one by one" from X9e to horus, but i feel comfortable, 'cause a lot of switches are used by scripts.
Nevertheless there is some work to rearrange switches & functions within oTx due to the "other ergonomic paradigm" of a horus, of course.
So I'd build a systematic "migration paper" which switch definitions in which area had to be changed from a X9e (source) to a horus(target) design.
I would have expected that an Input like "LS", which does exist on both TX's, wouldn't have changed by companion.
By now I've to adopt my paper to these "new" source inputs after companion X9e>>horus conversions, 
NO BIG DEAL !

Of course, it's beta software, and every user has to check all kind of migrations for plausibility.
In case there is "internal definition" that there is no must of a 1 by 1 mapping between X9e and horus inputs (cause it would exceed development work... or so on..) it's an understandable decision.
But if so (suggesting same behaviour by a standard X9e), a hint that there are expectable changes by selecting between Tx profiles would be useful (my guess)

3djc, 2017-04-22:
If your additional switches are 3pos to then yes, that data corruption should not happen. Having your 366 would be nice. 

Yes basic switches should go through and what you are experiencing should be resolved. At some point in time, we would like to have user defined transfer maps, but that's for 2.2.x

strgaltdel, 2017-04-22:
maybe a issue with my firewall, can't attach zip files here, any preferred source for filesharing withou need of registration ?

strgaltdel, 2017-04-22:
OK,try
http://www53.zippyshare.com/v/d24mEI2l/file.html

two files,
the "366" was uploaded within companion 366 (backup purpose),
the other one is an upload from companion 369 after FW 369 conversion.
Look at Radio setting, global functions...

kilrah, 2017-04-22:
www.filedropper.com

Even if both are "LS" they are still different things becasue the other controls make it so that they aren't in the same position in the list so the mapping isn't 1:1. 

Ideally all these cases would be handled but it takes motivation for someone to code that, with so many different platforms it starts to create a whole lot of individual cases.

strgaltdel, 2017-04-22:
Hi Kilrah, thanks again for explanation,
i was afraid of something like this.

as mentioned before, my suggestion would be to add one more line in the window which occurs by changing TX profile, in order to sensitize the user 
(obvisouly at least i wasn't aware of this, maybe the majority is)

Thanks and bye !

kilrah, 2017-04-22:
If a more complete conversion doesn't make its way in by the time we get to final release we'll put an additional warning. For now there's still hope some work is done on that.

strgaltdel, 2017-04-23:
it seems that the remaining "issue" is more a question of a consistant value presentation between Firmware and companion.

Cause the input "RS" controlled the loudspeaker volume, i named it "LS" in the config.
(OK, should 've been aware that "Vol" would've been better and would have avoid irritations)
[image]

By now, Tx firmware shows the "Input Hardware", 
[image]

and Companion shows the "Input label"
[image]

.. which is a change in behavior between companion 366 to 369.
personally i would prefer same results between firmware & companion.

3djc, 2017-04-23:
Oh, that's a nice bug :) Name should be displayed, but are not JUST for RS (LS works as it should)

strgaltdel, 2017-04-23:
Yes, all is running fine, just a "display feature" (-;

3djc, 2017-04-23:
and the same exist for S2 on x7, at least we are consistent !

PR #4843 (2017-04-23): fix drawSource
drawSource was not displaying name when defined for the last POT source. on x7 and x9

X12 does not seem to be exhibiting the issue

This fixes #4840
closed 2017-04-23 by bsongis