opentx/opentx #4841: Companion 2.2(n369) logical switch V1 name changes from Tx to companion

calril312, 2017-04-22:
LS30 & LS32 V1 were set to "JSy" on production Horus-X12S Tx but changed to "L1" when models and settings were loaded into companion.
[image]

1Sven, 2017-04-22:
I see this also as a problem. And you can not simulate the JSy/x on Companion becorse the stick is not mouveble. For test I'm using L1 or L2 that maks the confusion much bigger...
I hope of a probatly solution to!

bsongis, 2017-04-24:
Is this issue still there in the last RC?

kilrah, 2017-04-24:
Can't reproduce on n369 with setting done on a Horus also running on N369 and read into companion.

calril312, 2017-04-24:
I believe the model was created on n368 but the LS was created on n369. It was created on the radio, not companion or the simulator. Here is the "print model" file. What else can I provide that might help?
[DemonS8R 540.pdf](https://github.com/opentx/opentx/files/953486/DemonS8R.540.pdf)

kilrah, 2017-04-25:
zip the MODELS and RADIO folders ftom your radio's SD card and send us a link to that.

calril312, 2017-04-25:
[MODELS.zip](https://github.com/opentx/opentx/files/954922/MODELS.zip)
[RADIO.zip](https://github.com/opentx/opentx/files/954923/RADIO.zip)

kilrah, 2017-04-25:
OK, must be something with @mpaperno's source names - JSy takes the name set for L1 in the hardware page if there's one

bsongis, 2017-04-25:
I take it

kilrah, 2017-04-25:
i think it's in eeprominterface.cpp:555, should test for sticks only

calril312, 2017-04-25:
Cool, well let me know if there's any other info you need. Thank you guys for all that you do for the RC & OpenTX communities. Seriously, your time and efforts are greatly appreciated!

commit b0183efa (2017-04-25): Fixes #4841
closed 2017-04-25 by bsongis