opentx/opentx #4846: X7 Fw update error when trying to update recievers

michael-blue2, 2017-04-23:
I have tried to update 2 of my receivers without luck. After a few seconds that i select the .frt firmware and press update external device, i get an update error.(and there are no leds on the receiver flashing). firmware: N369.
After i borrowed an x9D plus from a friend with the same version(N369). We swapped sd cards( as the firmware was in my sd) and flash both of my receivers without problem using the same cable.
(my module bay is working fine as i am using an orange module to fly micros)

powercycle, 2017-04-23:
michael-blue2, your radio is an x9D+ as well?  That is the current assumption.

michael-blue2, 2017-04-23:
I have mentioned it on the title, i am sorry that i forgot to mention it again. Its the taranis Q X7

3djc, 2017-04-23:
If you have no pins on the bottom of your x7, you can use the module pins after removing the module, I do it daily.

If the is a connector at the bottom, then it has to be use since module pins does not work for flashing if bottom connector is present

kilrah, 2017-04-23:
Note that if you have the bottom connector pins you need to use that with a normal cable, not the modified one.

michael-blue2, 2017-04-23:
my pins are on the bottom, wha tdo you mean by normal cable and not modified one?

michael-blue2, 2017-04-23:
https://oscarliang.com/ctt/uploads/2016/05/flash-frsky-rx-firmware-x4r-sb-taranis-connection.jpg
i used the same connection with the x7 and the x9d. Working fine with x9d, and fw error on x7

michael-blue2, 2017-04-23:
http://cdna4.zoeysite.com/Adzpo594RQGDpLcjBynL1z/cache=expiry:31536000/resize=fit:max,width:1200/compress/http://s3.amazonaws.com/zcom-media/sites/a0i0L00000QyvG7QAJ/media/catalog/product/_/d/_dsc6145.jpg
Do you mean the pins on the left? of the bottom of the x7? yes i have those on my taranis, i will test it now from there and swap +- to be a normal futaba plug

3djc, 2017-04-23:
Please read what I wrote...

michael-blue2, 2017-04-23:
Thank you guy sfor your support, i got confused about bottom pins, i had no idea. And i tought it was a version error. Thanks again

closed 2017-04-23 by michael-blue2