opentx/opentx #4848: Companion & Horus - model files are getting corrupt by SD sync

dk7xe, 2017-04-24:
Companion - any version
TX - FrSky Horus

**Issue**
model files on TX are getting corrupt in case of SD card sync initiated before "Read models and Settings from Radio" is executed. Only the current selected model is ok still.
This happens especially when upgrading Companion to a newer version and doing the SD card sync before upgrading the TX for backup reasons.
The issue occured several times in the past with different versions of Companion & OpenTX. 

**Proposal**
Sync *.bin files only in one direction from TX to PC or exclude them from SD card at all.
And do update of *.bin files on TX via "Write Models and Settings to Radio" only.

kilrah, 2017-05-02:
What do you mean by "corrupt" exactly?
The only thing that could in theory happen is that if you opened your SD card copy in the simulator models from these are now newer than what's on the card and they get copied across.

dk7xe, 2017-05-02:
Corrupt means that oTX is not able to load them any longer. In the model selector the models are displayed as being invalid.

3djc, 2017-05-03:
This means that data has gotten damaged somehow during transport. It can be many things, an opentx issue (radio or companion side), a usb cable issue,, a pc issue,...

At this point in time, we are considering disabling the sd sync feature altogether for 2.2.0, because thats not something we want to rush before 2.2.0 release

projectkk2glider, 2017-05-05:
The issue #4848 might be related.

dk7xe, 2017-05-06:
No, Companion is not crashing at all in na case.

powercycle, 2017-05-07:
If I delete a file that isn't working on the SD, then choose sync, that file works as expected.
Not sure why the "write" portion of the code is giving us different results than the "copy" portion of the code.
If it does choose to write instead of copy, it will break the file.

Makes me wonder if QTextStream doesn't deal well with non-text files.

mpaperno, 2017-05-07:
Indeed, looking at [the code](https://github.com/opentx/opentx/blob/next/companion/src/process_sync.cpp#L115), if a destination file doesn't already exist then it simply copies it from source folder directly.  However if it does exist, _for some reason_ it tries to copy the file byte-by-byte, using QTextStream.  Which indeed is for... text, but none of the text-specific features are used anyway (there's a QDataStream for more generic content).

Does anyone know/remember why it does this?

Otherwise it's really basic code, just copies files/folders recursively... it should be easy to fix if the byte-level copy can be eliminated.

It could well be related to the crashing issue #4837 since that happens in a videos (binary data) folder.

mpaperno, 2017-05-07:
Reading more carefully, I guess it ties to do a byte-level compare to... make sure the contents actually changed before copying?  It's a nice idea but in practice probably takes more time than just copying (plus using strings is wrong).  A better way to compare would be to generate hashes and compare those.

`QCryptographicHash::hash(sourceFileData, QCryptographicHash::Md5) != QCryptographicHash::hash(destFileData, QCryptographicHash::Md5)`

3djc, 2017-05-07:
That's indeed quite odd. The only thing I can think off is in the early days, when sd write was really really slow, it could have made sense to try to confirm a fiel really need to be be overwritten, but nowaydays, it juts feel out of place

powercycle, 2017-05-07:
What about just deleting the file and then "copying" it as above?

kilrah, 2017-05-07:
> If I delete a file that isn't working on the SD, then choose sync, that file works as expected.

This is weird in the first place, a normal "sync" would be expected to delete the file on the other side in this case. It's kinda good it doesn't at this point, but it's still weird behavior.

mpaperno, 2017-05-07:
I'm on it...   got a bit carried away though...

[image]

The hash compare works great BTW.

jtaylor2, 2017-05-07:
I like it.  I never used sync because I was never sure what was going to be overwritten.

closed 2017-05-08 by bsongis

dk7xe, 2017-05-09:
Thank you guys!