opentx/opentx #4867: 6 character limit in voice file selection

boTy635, 2017-04-27:
I made a voice pack for my Taranis Q X7 with OpenTX Speaker, and all the files with more than 6 characters in the file name are not visible in the list at special functions>play track. They are also not visible in the global functions list. However, they are visible in the SDcard explorer menu.

kilrah, 2017-04-27:
Normal, keep length to 6 chars max.

closed 2017-04-27 by kilrah

boTy635, 2017-04-27:
Tried that, but the problem with that is that it doesn't recognize system sounds like the trim center sound.

Or should I only rename my custom files to 6 characters and not the system files?

kilrah, 2017-04-27:
Only custom, system filenames must not be touched.

boTy635, 2017-04-27:
@kilrah Okay, I will do that, thank you!