opentx/opentx #4879: [Cosmetic issue] Input are not displayed correctly

LapinFou, 2017-04-30:
With the N371, the inputs are not displayed properly.
It is `:Input` or `[Ix]:Input` instead `(input)` or `[Ix]Input`
[image]

mpaperno, 2017-04-30:
Why is this not "proper?"  The inputs were always displayed with the `[ ]` brackets (presumably to mimic the inverted appearance in the radio UI).  I added a colon to a) make it easier to read (especially in the selector lists) and b) make it more consistent with how the other sources are shown (eg. `CH01:Name`).  

The channels were never in brackets, but some separator was needed to show the name.  Parenthesis are a) extra characters (3, with the space) and b) sometimes harder to read because in proportional fonts they align very closely with the characters they surround (again, especially in selector lists).

I'm open to opinions and changing it to whatever most people prefer, but the above is my rationale for the changes.

Now, looking at the actual Inputs screen (Entrées) I do see that there in no colon in the left column.  So maybe one should be added there to make it more consistent.  Is that what you meant?

[image]

LapinFou, 2017-04-30:
Sorry, my explanation wasn't accurate enough.
What I mean is with the N371 the display in **Mixes** (not **Inputs**) is not as before.

kilrah, 2017-04-30:
It's not as before becasue he added the `:`...

mpaperno, 2017-04-30:
"There can be no progress without change."  :)   The question is, is it better this way?  

A colon is also used in the mixer lines to denote replacement, but IMHO it still looks OK (replacement could just be `=` but kinda late to change it now I guess).  

The other advantage of the changes is that there is only one place in the code where the channel (etc.) display names are defined (less to translate/maintain  ;-) ). Except now I see where I missed the input name line printer (for Inputs screen), so I should fix that to be consistent (either way).

kilrah, 2017-04-30:
I actually think it's clearer as it was before without the `:`...

mpaperno, 2017-04-30:
Even in the dropdown selector lists?

kilrah, 2017-04-30:
Yes, I don't see them add anything, just distracting..

mpaperno, 2017-04-30:
Personally I found it easier to read in some cases like with `D/E/R/P` etc. chars next to them.  Maybe I should start wearing glasses at the computer  :)

[image]

[image]

(And yes the window title is missing channel name, just noticed that.)

mpaperno, 2017-04-30:
Just for more example/comparison:

[image]

[image]

Also why `Input05` instead of just `[I5]`?

Oh and should it be `[I05]` or keep w/out leading zeros?  I kinda prefer the leading zero because it keep the width consistent (and we already do that now for many other numbered items in the UI).

So many options  :)

mpaperno, 2017-04-30:
[image]

kilrah, 2017-05-01:
I think the ] makes enough of a delimiter on the inputs/mixer pages it's all pretty clear - BUT in the dropdowns it's hard to read indeed. I'd just put a space there instead of the `:`. If you find the pages warrant one too that's fine as well.

>Also why Input05 instead of just [I5]? Oh and should it be [I05] or keep w/out leading zeros? I kinda prefer the leading zero because it keep the width consistent (and we already do that now for many other numbered items in the UI).

No idea - agree [I05] would be cleaner i.e. with the leading 0.

I do like the `:` in the Curve though since there's no other delimiter already. Maybe also for flight mode/GVAR?

LapinFou, 2017-05-01:
OK, I didn't understand it was done in purpose.
It is different, but it is more clear (I saw it right away, so the goal is aimed 😉 ).

I also like the leading 0. 👍

mpaperno, 2017-05-01:
I tried spaces but it's a bit ambiguous in some contexts, esp. in the input/mixer lists where everything is on one line.  The colon removes any ambiguity (I think).

Channel, curve, and GVars all follow the same format (eg. `GV01:Name`). (FM is the last item which doesn't display custom names... need to get around to that, but there's about a zillion different places right now where "FM##" is formatted/printed.)

I agree the colon in input names on the inputs/mixes screen isn't necessary for readability, but nice in other places.  I could really go either way on that, but I lean somewhat to consistency throughout the UI (and less exceptions in the code).  I could also see getting rid of the `[ ]` altogether, personally.  Icons like in the radio UIs would be ideal (actually that would be pretty easy to add at this point ;-)  ).

(whoops, thought I posted this a while ago, but never hit the green button  :-/ )

mpaperno, 2017-05-03:
RFC @ #4888

closed 2017-05-11 by mpaperno