opentx/opentx #4881: Possible incorrect external PXX pulses

MikeBland, 2017-04-30:
In pxx_arm.cpp, the pulse duration is set to either 48 or 32. This value gets put into the ARR register of the timer (via DMA). Counters count from 0 to the ARR value, then go back to 0. This will result in either 49 or 33 counts for the pulse width, thus making the pulses 0.5uS too long.
The "rest" value is calculated using one less than the duration value, so the total period will also be incorrect.
The duration should be set to 47 or 31, with the "rest" value being decremented by (duration+1).

bsongis, 2017-04-30:
I am 99% sure I checked with the logic analyser the pulses length, but this needs to be double checked

schwabe, 2017-04-30:
With my logic analyzer I am get 8 µs for the high level of pxx external and 8,5 µs and 16,5 µs for the low level of the signal. I don't have a PXX documentation so no idea what it is supposed to be.

kilrah, 2017-04-30:
Then @MikeBland is right, supposed to be 8 and 16

schwabe, 2017-04-30:
[Uploading pxx external.logicdata.txt…](the capture)

bsongis, 2017-04-30:
Then this needs to be fixed before 2.2.0 is out, @schwabe would you fix it? If not just fire a post on the chat, I will take it.

closed 2017-04-30 by bsongis

kilrah, 2017-05-01:
Just to clarify this had no influence at all, the XJT accepts several us deviation from the spec. Even FrSky's OS on the Horus doesn't give "accurate to spec" output.

commit f525b176 (2017-04-30): Fix #4881 (#4884)

Thanks Mike for noticing: In pxx_arm.cpp, the pulse duration is set to either 48 or 32. This value gets put into the ARR register of the timer (via DMA). Counters count from 0 to the ARR value, then go back to 0. This will result in either 49 or 33 counts for the pulse width, thus making the pulses 0.5uS too long.

Tested only with external PXX