opentx/opentx #4885: N371 - X9D+ - Hello.wav does not play on start-up

powercycle, 2017-05-01:
Just upgraded from N362.  
Downloaded X9 zip
Synced SD using N371 version of companion.
hello.wav does exist on card but does not play.
This seems related to having existing files on the SD card or some other problem related to SD card.
Clearing the SYSTEM directory on the SD, then resyncing fixed the issue.
Needs more testing.

kilrah, 2017-05-01:
Most likely duplicate of #4848, SD sync has issues. Just fix things straight on the card for now.

closed 2017-05-01 by kilrah

powercycle, 2017-05-02:
That is what I did.  I'll see if I can find a way to "break it" consistently.  Hopefully narrow down the sync issue.
Unless of course someone already knows what the problem is and just hasn't had time to fix it.

kilrah, 2017-05-02:
Nope, it isn't clear what the problem is.