opentx/opentx #4887: Logic switch problems with 371

tomkfly, 2017-05-01:
Using a logic switch to operate a mix only works once, without restarting radio or simulator, both Companion and Horus. Also using Amber 2.2 pack some sounds don't work in Sim., but do in radio.
(Knfdg) is one.
Another niggle, in Simulator if the width of the monitor panel is reduced so it fits alongside the Horus panel and not part cover it, the Logic switch names are too long for the button width.

kilrah, 2017-05-01:
That's.... just not possible. 
Please post exact steps to reproduce and your setup files.

tomkfly, 2017-05-01:
The problem with the Knife Edge sound not working in the Companion Sim has gone away. It just started working. It was working on the Horus and worked in Special Functions using the play button, but on the Companion Sim The Lsw indicator showed the switch was on, but no sound was produced. All the other voices, I was using worked OK. Now, without any further adjustments it has started to work???

I have a setup to initiate Self Check on an S6R it uses Lsw10, as the switch for Ch12, which is setup with AND function and uses the not "!" operator of the Lsw that releases throttle safety and SC↓. It uses a curve to produce the 3 required operations of to and away from zero. It works once and wont work again until the Sim or Horus are cycled off and on again. I've checked through everything numerous times and can't see anything where I might be missing something.
Dropbox link to my .otx file
https://www.dropbox.com/s/0dfkkbpzndzbidz/LS%20problem.otx?dl=0

Dropbox link to Image showing the Lsw names cramped when the monitor is reduced in width to fit alongside the Sim. Not a major problem. 
https://www.dropbox.com/s/2uq6wiv5hkja5zq/LS%20names%20crammped.PNG?dl=0

closed 2017-05-01 by tomkfly

mpaperno, 2017-05-01:
No problem, glad you figured it out.  For the sound issue, perhaps check the simulator volume gain setting in Companion Settings.  The LSw display issue has already been fixed (#4877).

tomkfly, 2017-05-01:
I have sett the Volume gain to 3.0, but it didn't make any difference. It's really a strange one, most sounds work okay. The Knife edge one is the only one I'm using that I have the problem with and that worked for a short time this evening. In trying different things out I did come across a couple of others that wouldn't work.