opentx/opentx #4892: Chars in fonts not following ASCII std, not consistent across sizes, width not consistent

floaledm, 2017-05-03:
I'm sure there are good reasons for some of my observations (I looked at SMLSIZE, std size (0) and MIDSIZE), but I'd like to discuss to see if there are changes that could be made as a result, so here it is:

1) Chars in fonts not following ASCII std:
Some characters are represented with a symbol that does not match the ASCII standard. For example, @ is displayed as °.

2) Chars not consistent across sizes
In std size font, # is displayed as a check mark but is displayed as # on some, if not all, other fonts.

3) Chars width not consistent
The width of at least a few characters in the SMLSIZE font (04x06) are 5 pixels wide (e.g., #, $, +), but shouldn't all characters in that font be 4 pixels wide or less?

The code I used to make those observations:
```lua
local str = "!\"#$%&'()*+,-./:;<=>?@[\\]^_`{|}~"
for i=1,#str do
    local pos = i*6-5
    local character = string.sub(str,i,i)
    lcd.drawText(pos, 10, character, SMLSIZE)
    lcd.drawText(pos, 20, character, 0)
end
```
The resulting screenshot:
[image]

kilrah, 2017-05-03:
That's normal and has roots all the way back to the days of the 9X platform to save resources. So deeply ingrained that the entire text layer would need to be redone so it's unlikely to change anytime soon, at least not before AVR support is dropped.

floaledm, 2017-05-04:
Thanks for explaining that! So does the AVR platforms use only one font file to save memory, and therefore special characters required on those displays needed to replace some std ASCII characters in that font file?

I might not understand this fully since I'm not familiar enough with that part of the code, but since there's a font_05x07_avr.png file and a font_05x07.png file, I assume that the AVR platforms use that first font file, and the other platforms the second file possibly complemented by the "extra" font file... Is that not the case?

closed 2017-11-08 by kilrah