opentx/opentx #4893: QX7: Slow menus and voices. Channels don't respond.

miginv, 2017-05-03:
Since I updated to 2.2.0N369 with Spanish menus, after some normal operation, Taranis Q X7 started to show a very odd behavior:
- Voices seem to be played very slow so it is impossible to understand them.
- Menus are slow.
- Radio transmitter and receiver are binded since receiver led becomes green when on and transmitter vibrates when receiver is off. However, channels updates are not displayed in the Cleanflight configurator. They have fixed values besides I move all sticks and switches.
I have done this troubleshooting:
- Uploading different firmwares (both Opentx and bootloader): 2.2.0N364, 2.2.0N370, 2.2.0N371 and from Frsky website (http://www.frsky-rc.com/download/files/Firmware/FIRMWARE-X7_170213.zip)
- Removing SD card and using different sdcards.
These are some videos that shows the behavior: 
https://www.youtube.com/watch?v=DoTi1rIMcv4
https://www.youtube.com/watch?v=UvzA9gl8svg

Thanks in advance.

kilrah, 2017-05-03:
> Uploading different firmwares (both Opentx and bootloader): 2.2.0N364, 2.2.0N370, 2.2.0N371 and from Frsky website (http://www.frsky-rc.com/download/files/Firmware/FIRMWARE-X7_170213.zip)
Removing SD card and using different sdcards.

And, did it change anything?

Looks like a hardware problem with the main crystal/oscillator.

3djc, 2017-05-04:
Yeah looks like your radio needs to be returned for warranty.

miginv, 2017-05-04:
Thanks for the answers. Changing firmwares and removing the SD card didn't change anything. So it seems that the mainboard (with the sd card reader, usb, sticks connectors, etc) has something wrong, doesn't it?

kilrah, 2017-05-04:
Correct.

closed 2017-05-04 by kilrah

projectkk2glider, 2017-05-04:
Unless you think you changed firmware, but you actually did not. You should verify the actual flashed version from the radio setup menu to see if you actually succeeded to flash another firmware.

miginv, 2017-05-04:
Perhaps I am missing something. I uploaded the bootloaders versions from Opentx and the Opentxs versions from the bootloader. What I couldn't is to enter into the DFU mode besides I followed the Zadig manual (I tried in Windows 10 and 7). I also tried dfu-util from Mac and Linux, but I never get the stm32 recognized. Before the issue the DRL simulator recognized the radio but now it doesn't, so the problems also affects the USB. Anyway, if the DFU is not modified by upgrading the opentx itself or opentx bootloader, it shouldn't matter.
According to these pictures I think I actually changed the firmware, but please, can you confirm?

[image]
[image]
[image]
[image]
[image]

kilrah, 2017-05-04:
Yes all good. 

A wrong system clock would cause everthing you see (USB can't work if clock is incorrect, internal module won't recognise the input signal that's highly timing-dependent explaining the lack of response from the receiver, etc...)

miginv, 2017-05-04:
It makes sense, but then is a SW or HW issue?

kilrah, 2017-05-04:
Hardware, if it was software nobody would have an X7 that works.