opentx/opentx #4894: Trainer Simulator Screen

Waldo-P-Barnstormer, 2017-05-05:
Taranis X9D		Companion      2.2 N371 RC18  
Setting up a trainer configuration where the throttle on the students Tx must be down or transfer from instructor to student won't occur. Thought I had it correct but it wouldn't work in Companion. Then I lowered the R/H stick on the Trainer Simulator screen and it worked. I then noticed that all stick functions were crossed on that screen.  Rudder and Throttle on R/H, Ailerons and Elevator on L/H stick.
 I fly Mode 2.
 
Decided to try it on the radio and it works properly there. The student's radio has to have the throttle (L/H stick) down before the instructor's radio can pass control. All controls functioned properly after.
It looks like the stick positions are crossed in the Trainer Simulator screen.
Jim

kilrah, 2017-05-05:
Are you sure you set the mode right in the student file 's general settings?

mpaperno, 2017-05-06:
There is no setup for the trainer simulator right now, it doesn't know anything about the control mode or which channel is which on the student radio. So this could be an enhancement.  For now the radio trainer settings needs to be adjusted to match: 
```
R = CH1
E = CH3
T = CH2
A = CH4
```

Waldo-P-Barnstormer, 2017-05-06:
OK, coming from Futaba three years ago I stayed with the AETR channel order. That explains the issue. It works on the radio so will keep it as is, especially now that my trainer has the S6R installed and it needs the AETR channel order to work. Thanks for the explanation and all the effort you guys put into OpenTX.
Jim

closed 2019-11-15 by stale[bot]