opentx/opentx #4903: New feature wanted, posibility to do bitwise math on telemetry

kajsilander78, 2017-05-07:
Hi!

I would like to have a possibility to do some math operations on telemetry...
I have a naze32 with x4rsb sending telemetry, one of the values I get is Tmp1.
Tmp1 : Status of FC, sent as 4 digits. Number is sent as 12345.
Digit nr 4 is the armed status (1=Ready to arm, 2=Arm prevented, 5=Armed). I'd like to be able to determine value of the digits individually to make taranis tell actual status of FC....
(Binary 8 AND Tmp1, if the digits are 8bits?) Or is there other methods?
Hope you understand
what I'm looking for...
/Kaj

3djc, 2017-05-07:
You have Lua for things like that.

mpaperno, 2017-05-07:
I understand the need and usefulness of sending bit-encoded data for "custom" telemetry (I do the same thing), but I'm not sure what kind of interface would help?  Unless it was specific to each system (naze, APM, etc)... but that's not realistic for OTx.  What did you have in mind?

At one point (couple years ago) I made some changes to the telemetry simulator to allow hexadecimal base input, but that got removed in 2.1 (I think) in favor of the decimal-only spin boxes.  Hex is much easier to deal with for bitfields, IMO.  I have plans to restore that functionality, and improve on it (the whole telemetry simulator UI is getting a revamp and will be much more useful).  Like keeping a list of previously-entered values for each sensor.

Regardless, you'll need to use some external tool to calculate the actual bitfield values that your FC telemetry sends.  Eg. I often use Windows' built-in Calculator, which has a great "Programmer" mode.   Or ask your FC programmer to add some calculation features in their GCS/setup GUI.  😉

schwabe, 2017-05-07:
I not sure where you want the hex to be displayed. We could add Raw Hex in addition to Raw to show a sensor in hexadecimal instead of decimal. Showing binary is not going to fit as telemetry values are 32 bit. For any more advanced as @3djc there is lua.

mpaperno, 2017-05-07:
I might have totally misunderstood what the OP was asking.  I thought it was about simulating the telemetry values to test scripts.  Re-reading it again, I'm not sure where I got that from :)

kajsilander78, 2017-05-07:
Hex would work great with some possibilities to filter out (AND, XOR...) with custom values to filter  out the info wanted. Is it possiible to duplicate sensors? Thinking of diffrent AND-mask to get diffrent info on separata "sensors". Another way to deal with with it would be a new "math screen" where you could have some basic math functions to manipulate variables into new... Yeah and there is LUA, anybody use it? ;) Would be handy to do simple stuff without involving scripting... 
/kaj

kajsilander78, 2017-05-07:
Haha, hope it got readable! Writing on a phone with txt-correction to Swedish without glasses in bed :)

mpaperno, 2017-05-07:
Well there is already a provision for formulas, and there's also a bitfield unit type defined (though it's "hidden" right now I believe, only for R-Bus use?).  And you can have the same sensor multiple times in a formula.  So that's something.

I'm curious what would you do with a resulting value though, w/out scripting?  Use it as input for LS/SF?

What you describe would be fairly trivial in a Lua script... there are even bit operators already available.  Then output whatever values you need from a background "mixer" script.  To me that sounds easier than dealing with some obscure formulas in a limited GUI (I can't even keep the LS functions straight).

kajsilander78, 2017-05-07:
Oh, I see now that my issue wasn't really clear...

In telemetry screen I have a sensor called Tmp1.
Tmp1 is a five digit number 12345.
Each nr are different status numbers for the flight controller. 

Example: 
Digit nr 5=status of arming the quadcopter.
Different values for digit 5 are presented
by FC.

1:Unarmed
2:Unable to arm
5:Armed

Just missing the feature to get rid of digits 1-4 and keep the digit for evaluating the "arm status digit" for all kind of stuff. (For example have taranis to tell me if quad is armed/unarmed). As It is now taranis just tells me that I have switched a key for arming, no verification if it actually is armed(or disarmed).

kajsilander78, 2017-05-07:
The other digits are other info, GPS status, flight mode and so on..

mpaperno, 2017-05-07:
OK, that's an interesting way to do it, but I'm not sure it changes much (you still can use math/bit-ops).  Except that in this case you could also just use simple split/array functions to extract the data (in Lua, natch) .  :)

kajsilander78, 2017-05-08:
OK, I'll give up :) Guess I'll have to do it in LUA.

closed 2017-05-08 by kajsilander78

3djc, 2017-05-08:
Oh, I remember working exactly with this, Bits operations would not help you much since it is bit within digits within a number and cannot be solved with simple bit masking.

What can be done is an LUA that reads and parse those value, and create virtual sensors accordingly. There is something similar done here of redundancy bus decoding : https://github.com/opentx/opentx/blob/3djc/rb-decoder/radio/sdcard/horus/SCRIPTS/MIXES/rb_dec.lua