opentx/opentx #4904: Companion 2.2 can't set custom failsafe to Hold or No Pulse

jarhart, 2017-05-07:
Companion 2.2 RC18 (N371), Custom failsafe mode setup: the dropdowns for setting the positions to `Hold` or `No Pulse` don't seem to change the model and it always gets reset back to `Value`.

To reproduce:
1. Open a "models and settings" file or import from a TX.
2. Open one of the models.
3. Set the failsafe mode to `Custom`.
4. Set one of the Failsafe Positions to `Hold` or `No Pulse`.
5. Close and re-open the model.
6. The Failsafe Position is `Value` instead of whatever it was set to in step 4.

Workaround:
Set failsafe in the TX after writing Models and Settings from Companion.

kilrah, 2017-05-07:
Radio type?

jarhart, 2017-05-07:
Tarans and Horus

mpaperno, 2017-05-07:
Confirmed.

commit ca25696d (2017-05-08): [Companion] Radio module setup: Fix custom failsafe no-pulse and hold options not being saved properly (closes #4904).
closed 2017-05-08 by mpaperno