opentx/opentx #4905: "FrSky D (cable)" telemetry not working on 9xr-pro

rodericktaylor, 2017-05-08:
"next" branch (2.2) built by me from commit e2502772cbaee573684fef0556d0174fbf473b0f
9xr-pro, using version 9 SD card contents

"FrSky D (cable)" telemetry not working with an FrSky DJT module.  It's not a hardware issue, as installed Ersky9x (release 220, latest) and it works fine (ersky9x refers to the cable conection as com2).

I can't test "FrSky D", as I only have a DJT.

rodericktaylor, 2017-05-08:
I managed to decipher what was going on.  The taranis/horus code seems to have moved things on a bit and sky9x (9xr-pro) hasn't caught up.  I've hacked serial2 support back and it seems to work now.  I'll see if I can help the sky9x catch up a little.  If not I'll work out how to share my fix and do so.

guilopesn, 2018-05-28:
Is this milestone active?

rodericktaylor, 2018-05-29:
merged in #5220

closed 2018-05-29 by rodericktaylor