opentx/opentx #4915: Download firmware

rcabuser, 2017-05-11:
I've tried several times to download firmware/software, but when trying the download link for the given firmware/software i get the message "site can't be reached" I tried on a windows 10 computer with all the different browser's, and i tried with a Mac but no luck.

3djc, 2017-05-11:
companion version ? radio type ?

rcabuser, 2017-05-11:
I tried to download the wizard lua script

3djc, 2017-05-11:
What script are you refering to ???

rcabuser, 2017-05-11:
model wizard

3djc, 2017-05-11:
what model wizard ? what link are you following ? what instructions are you followinig ?

rcabuser, 2017-05-11:
[image]

3djc, 2017-05-11:
ok, that starts helping

3djc, 2017-05-11:
Fixed

closed 2017-05-11 by 3djc

rcabuser, 2017-05-11:
Ahh great, thx man:-)

 
Closed #4915.