opentx/opentx #4917: Companion 2.2 Model edit changed names not immediately updating across drop downs

elecpower, 2017-05-11:
gvars, curves, etc name updates not immediately reflected in the new style type:name drop downs. Need to close and reopen the edit dialogue for the refresh to occur.

Updates okay for Inputs and Mixes sub dialogues.

Latest local Ubuntu 16.4 compile for Taranis X9D+

mpaperno, 2017-05-11:
That's correct, in some cases there's caching going on right now to improve overall performance.  I have new code with better updates already working here, but it will most likely need to wait for 2.2.1 (to avoid risk of breaking anything else for 2.2 release).

elecpower, 2017-05-11:
@mpaperno yeah 2.2 from what I have see has been full on for the dev team and needs to get stability to release. The issue is just an annoyance but thought I would raise. Not surprised you have already been working on it.

kilrah, 2018-07-10:
RC4 still (or again?) doesn't update the source dropdowns on the LS page when an LS is added.
AKA if you set L01, you won't be able to use it as in L02 before closing/reopening the window.

kilrah, 2018-07-10:
Actually it depends on the function. If you select OR in L02 the resulting source list does update and contain it. If you select a>b it doesn't. 
LS's being available in an a>b function is actually irrelevant and it would make sense disallowing them there, but they were available before so might not want to remove it altogether...

mpaperno, 2018-07-10:
Yes, I see.... the Sources lists aren't updating, just the Switches lists are.  Investigating...

mpaperno, 2018-07-10:
Well I don't love the workaround (works but inefficient), but I think a proper refactoring should wait for 2.3.

PR #5936 (2018-06-03): Fix remaining model editor sync issues with RawSource changes.
Fixes RawSource updates in Custom Function and Heli pages selectors.
Closes #4917 
Closes #5933
commit 3af2b090 (2018-06-03): Fix remaining model editor sync issues with RawSource changes. (#5936)

* [Companion] Fix RawSource updates in Custom Function selectors (part fix for #4917).

* [Companion] Fix RawSource updates in HeliPanel selectors (closes #4917).
commit 39db1b65 (2018-07-10): [Companion][LogicalSwitchesPanel] Workaround for added LSs missing from Sources selector on LS edit page (closes #4917 (again)).
closed 2018-07-10 by mpaperno